swift-nav/piksi_firmware #223: console.py fail on linux: traits.etsconfig.api missing

ggregory8, 2015-01-10:
Hi guys,

I have just received my Piksi RTK kit and am trying to get the console up and running on Linux. I have followed instructions as per http://docs.swiftnav.com/wiki/HOW-TO:_Running_the_Piksi_Console_from_source and when running `python console.py` I am getting the following error message 'Traceback (most recent call last):
`  File "console.py", line 43, in <module>
    from traits.etsconfig.api import ETSConfig
ImportError: No module named traits.etsconfig.api`

I am running Ubuntu 13.04. Any idea whats going on?

Thanks.

ggregory8, 2015-01-10:
Sorry guys I just found the swift-nav mailing list. I would have checked here before posting here. I found a similar issue on windows:
https://groups.google.com/forum/#!searchin/swiftnav-discuss/trait/swiftnav-discuss/tVioTXgpXtQ/JhtWMDz74rcJ

It appears I have dependancies missing. These should be added to the instructions. I have installed the following
`pip install traits`
`pip install IntelHex`
`pip install chaco`
This chaco is failing with
`unable to execute swig: No such file or directory`
`    error: command 'swig' failed with exit status 1`
`    Complete output from command python setup.py egg_info:`

mookerji, 2015-01-11:
Hi @ggregory8, under `Installing the Prerequisites`, you're going to want to checkout this page: http://docs.swiftnav.com/wiki/HOW-TO:_Setting_up_developer_tools_for_Piksi . There's a setup script runnable from master (`bash setup.sh -x install`) that installs many of the dependencies. 

Let us know if you run into any problems?

Thanks,
Buro

mookerji, 2015-01-12:
... and I've done a bit of testing for Ubuntu 13.*, once the referenced PR is merged, you should be able to use it.

PR #224 (2015-01-21): Updates to setup.sh.
Support for Ubuntu 13.04 and permissions bugfix.

Closes #223 and #220.
closed 2015-01-21 by mookerji