swift-nav/piksi_firmware #269: Simulation time handling is funky

fnoble, 2015-03-17:
Right now the simulator measures time using `chTimeNow()`. This allows the solution time to drift with respect to the simulated GPS epoch.

Should be possible for the simulation to keep track of time exactly, but need to be careful of edge cases e.g. when solution frequency changes.

denniszollo, 2015-03-17:
But what if you are indoors and have no access to GPS time?  The system should resort to freewheel in that case I'd say.

fnoble, 2015-03-17:
Yeah, sorry I wasn't explaining well. It does and should freewheel, but right now the times reported in the simulated solutions are a little bit off, they should be exactly on the second but there is some offset of up to 100s of milliseconds.