swift-nav/piksi_firmware #284: Console is throwing FutureWarning errors from Chaco

mfine, 2015-03-24:
[image]

Seeing traces from the Chaco library in the console log:

```
/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py:95: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._data = newdata
Number of tracking channels changed to 11
/usr/local/lib/python2.7/site-packages/chaco/legend.py:424: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._cached_label_sizes = label_sizes
/usr/local/lib/python2.7/site-packages/chaco/legend.py:425: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._cached_label_positions = zeros_like(label_sizes)
/usr/local/lib/python2.7/site-packages/chaco/axis.py:498: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  for tickpos in mapped_tick_positions]))
/usr/local/lib/python2.7/site-packages/chaco/axis.py:499: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._tick_label_list = tick_list
/usr/local/lib/python2.7/site-packages/chaco/axis.py:500: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._tick_label_positions = self._tick_positions
/usr/local/lib/python2.7/site-packages/chaco/legend.py:478: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._cached_label_sizes = None
/usr/local/lib/python2.7/site-packages/chaco/legend.py:481: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._cached_label_positions = None
```

/cc @denniszollo

mfine, 2015-03-24:
Verified that this suppression quieted the errors in `console.py`:

```python
import warnings
warnings.simplefilter(action = "ignore", category = FutureWarning)
```

Good idea?

denniszollo, 2015-03-24:
We could think about putting the supression only on the line that produces error as to minimize collartoral

cbeighley, 2015-03-26:
This is the best issue I have ever seen.

mfine, 2015-04-10:
Fixed (for now) by https://github.com/swift-nav/piksi_tools/pull/13.

closed 2015-04-10 by mfine