swift-nav/piksi_firmware #297: Mask satellite feature

denniszollo, 2015-03-30:
To be used in HITL to simulate in-the-field adding and dropping of sats with the real roof antenna

denniszollo, 2015-04-13:
After sprint planning the thinking is that this would be a PRN mask, as opposed to something that is a one-off for testing.

henryhallam, 2015-04-28:
As well as permanently enabling/disabling one or more PRNs, there are a couple of further features that would be useful to have for debugging:
1. Mask at different stages, e.g. acquisition, tracking, nav bit sync, ephemeris decode, obs transmission, use in single-ended PVT, use in IAR filter, use in RTK baseline calc
2. Two modes: 
 a. Mask permanently - presumably via "setting"
 b. Momentary "channel reset" at any of those stages (where appropriate) - presumably via a new message type.

gsmcmullin, 2015-04-28:
I was going with the new message approach even for masking the satellites.  The message would just contain a prn and a disable flag.  If you set the flag the satellite would be masked from future acquisitions, and if currently being tracked, it would be dropped.

I'll extend this to a bitfield of flags and implement the flags for acquisition and tracking, the others can come later.

I don't think a setting is appropriate as I don't conceive any reason why you would want this mask to persist over a reset.

henryhallam, 2015-04-28:
Sounds good.

> I don't think a setting is appropriate as I don't conceive any reason why you would want this mask to persist over a reset.

The only circumstance I can think of where that would be appropriate would be another case like [SVN-49 / PRN01](http://www.positim.com/gnss/2009/05/detailed-look-at-gps-satellite-svn.html) where a satellite is broadcasting that it's healthy, but the signal isn't really usable for RTK purposes.  I think they'll be unlikely to do that again, though.

closed 2015-07-06 by gsmcmullin