swift-nav/piksi_firmware #322: Check ARM toolchain version at build time and setup time

henryhallam, 2015-04-07:
cc @mookerji 
* Store the intended version of the arm-none-eabi-gcc toolchain somewhere in the repo
* Check it in the Makefile
* Also check it in setup.sh
* If not installed, install it (setup.sh) or suggest how to install it (Makefile)
* If incorrect version installed, fail build or setup.  Suggest user uninstalls old version (suggest stow, etc for managing multiple versions).  Allow override flag.

henryhallam, 2015-04-07:
cc @gsmcmullin @fnoble

mookerji, 2015-04-13:
I might add, I think that we should probably just mirror a fixed version, and point all Linux and OS X installs for building the firmware to that version. This would alleviate the hassle around managing this dependency.