swift-nav/piksi_firmware #329: Implement watchdog timer

henryhallam, 2015-04-07:
Use Window Watchdog.  In EWI, print error message before hardware reboots.

Open: which thread should clear watchdog?

henryhallam, 2015-04-15:
/cc @fnoble @gsmcmullin Comments requested before I get too far with implementation...

Rationale: As part of general fault handling cleanup, we want to (1) know when and why a crash occurred, and (2) not leave the system permanently in a crashed state.  Using the hardware watchdog timer is the most foolproof way of accomplishing the latter, but there are some subtleties.

The simplest option is to clear the WDT in e.g. the main loop or the heartbeat (system monitor) thread.  That will catch certain kinds of lockup but not others, where one of the other threads might spin forever but Chibi will preempt it and allow the heartbeat thread to keep clearing the WDT.  So instead we want the WDT-clearing thread to require some kind of affirmative signal from each system thread, and check that all those signals are present before clearing the WDT.

My current plan is to use ChibiOS Events, in a similar manner to that [discussed here](http://forum.chibios.org/phpbb/viewtopic.php?f=2&t=2394), where they are applied to exactly the same situation.  The heartbeat thread loop will be the only place that the hardware WDT is cleared, and that loop will chEvtWaitAllTimeout() on the flags from each thread before doing so.   Under normal conditions that syscall will return immediately, because the heartbeat loop period is longer than the period of all the events we want to watch over.  If any of the other threads are locked up, it will return after a timeout, which we can detect and use to report which thread is the problem.

The STM32F40x has two watchdog timer peripherals, the Independent Watchdog (IWDG) and the Window Watchdog (WWDG).  The WWDG has the rather attractive feature of optionally triggering an interrupt shortly before resetting the chip upon timer expiry, which could be used to report that a watchdog reset is about to occur (and possibly what caused it).  Unfortunately there are two drawbacks to the WWDG: 1. It runs from the APB1 clock, which is ultimately derived from the front end (via the FPGA).  A fault in the front end configuration or in the FPGA could cause this to stop, and the watchdog would never be triggered.  2. It has a max period of 64 ms.  Events we'd like to watch over may occur less frequently than that (e.g. nav solution epochs), thus we'd need complicated logic in the watchdog thread which would somewhat defeat the objective.

Instead we'll use the IWDG, which runs from a dedicated internal RC source and stays active even if the main clock stops.  The period is configurable between 0.125 ms and 32.8 seconds; I figure we'll choose a period around 1 second.

At boot time we'll check the reset register RCC_CSR for presence of the IWDGRSTF which indicates that the reset was caused by a watchdog timeout, and report that via log_error().

fnoble, 2015-04-15:
Sounds like a plan!

fnoble, 2015-04-15:
Also, maybe paste a link to this into design-snippets?

gsmcmullin, 2015-04-15:
That sounds sensible.  I would have taken the simpler (lazier?) approach of just kicking the watchdog from the idle (lowest priority) thread.  If any other thread is spinning it won't get run.

I would also like a setting to disable the watchdog.  Having the option of connecting the debugger to a spinning Piksi is useful for debugging these issues.  It may also be useful to have another timer that will hit us with a breakpoint instruction just before we expect to get zapped by the watchdog.

closed 2015-04-27 by fnoble