swift-nav/piksi_firmware #336: Observation message week number is not modulu 1024 as the specification suggests

denniszollo, 2015-04-10:
Example obsevation packet shows weeknumber 1839 in the file: http://zazu.ci.swiftnav.com/job/piksi_firmware_Log_Analysis/12264/artifact/serial_link_log_20150409-164416.log

Our specification (and the GPS specification) says that the week number is supposed to be modulu 1024, which should render a week number of 1839 impossible.

fnoble, 2015-04-10:
Our documentation must be wrong. We correct for the GPS week number rollover by adding 1024 so the number we use internally is the true week number rather than the mod 1024 value transmitted in the ephemeris. We must remember to update in 2019 when it rolls over again!

fnoble, 2015-06-25:
Not a bug, closing

closed 2015-06-25 by fnoble