swift-nav/piksi_firmware #341: Robustly handle loding settings.yaml

mfine, 2015-04-13:
The settings.yaml loading is hard-coded and dependent on the location where the console is run from. When not run out of the `piski_firmware/scripts` directory, the console will throw the following exception:

```python
Traceback (most recent call last):
  File "scripts/console.py", line 256, in __init__
    SettingsView(self.link, settings_read_finished_functions)
  File "/Users/mark/swift/piksi_firmware/scripts/settings_view.py", line 303, in __init__
    self.settings_yaml = SettingsList("settings.yaml")
  File "/Users/mark/swift/piksi_firmware/scripts/settings_list.py", line 40, in __init__
    stram = open(path_to_file, "r")
IOError: [Errno 2] No such file or directory: 'settings.yaml'
```

The settings view should if possible be tolerant of settings.yaml loading exceptions and/or use a more robust file location scheme.

denniszollo, 2015-04-13:
Agreed, I will update the pull request #289 where I fixed this a few weeks ago on Piksi tools and Firmware

mfine, 2015-04-17:
This should be resolved now.

closed 2015-04-17 by mfine