swift-nav/piksi_firmware #360: Use a message whitelist rather than mask

henryhallam, 2015-04-14:
Per [discussion](https://github.com/swift-nav/libsbp/pull/102#issuecomment-93009109), let's use a list of desired output messages rather than the mask system.

As well as allowing full customization, we should include a few sensible default lists, e.g. "Default navigation output", "Minimal baseline output", "Default base station to rover", "Minimum data base station to rover".  These defaults could either live in Piksi or in Console.  They could also be specified with some additional field, e.g. 'message_groups', in the sbp yaml.

cc @fnoble @cbeighley @denniszollo @mookerji

fnoble, 2015-04-14:
Perhaps we could have the behavior that is the list is empty then it counts as let everything through?

denniszollo, 2015-04-14:
What about frequency?