swift-nav/piksi_firmware #370: Remove init known baseline feature from console

cbeighley, 2015-04-16:
Doesn't have that much value and has a lot of anti-value. For example, see:
https://groups.google.com/forum/#!topic/swiftnav-discuss/UpXC7QTub34

henryhallam, 2015-04-16:
Shouldn't this be in piksi_tools?

fnoble, 2015-04-16:
I think the problem isn't the feature itself so much as the fact that it currently inits the integer solution but not the KF. It would be way more robust if it simply inited the KF too.

I still think its a useful feature but maybe we shouldn't advise people to use it until this is fixed.

henryhallam, 2015-04-16:
@fnoble How much work is it to have it init the KF?  Sounds pretty straightforward and it would help a lot of people.  I'd be inclined to make that a priority for the next sprint if not sooner.