swift-nav/piksi_firmware #402: Investigate hardfault error

denniszollo, 2015-05-14:
We have received reports from customer about receiving this error:
ERROR: HardFaultVector : HFSR=40000000 CFSR=00001400 MMFAR=E000ED34 BFAR=E000ED38

we need to investigate and add a test to HITL potentially

gsmcmullin, 2015-06-18:
A significant number of hard faults (hopefully all) have been caused by corrupt flash.  The flash corruption was occurring during the bootloading process as a result of SBP message ID ambiguity and the two Piksis being connected during the process.

On completion of a flash command, the Piksi bootloader will send out MSG_FLASH_DONE on all ports.  The Piksi running the standard firmware interprets this as a MSG_FLASH_PROGRAM (same ID).  There is no check on the message length, so depending on what junk is in the buffer, a random chunk of flash may get written over.

This is fixed in #432 by removing the flash callbacks from the firmware.
There will be unpredictable results if both Piksis are bootloaded simultaneously.

cbeighley, 2015-06-18:
Both Piksies being bootloaded simultaneously should be okay, we fixed this in piksi_bootloader v1.2.

henryhallam, 2015-06-18:
Wow, that's pretty special.  Nice sleuthing.  @denniszollo do we know whether the customers who reported the issue were bootloading with connected Piksis?

gsmcmullin, 2015-06-25:
There don't appear to be any remaining hard faults after #432 was merged.

closed 2015-06-25 by gsmcmullin

denniszollo, 2015-06-26:
No one seems to remember how or when they bootloaded, so we only have our
best guess whether customer issues were caused by this.