swift-nav/piksi_firmware #404: Nap firmware check upon bootup can leave device unresponsive if NAP is out of date

denniszollo, 2015-05-15:
in main, if the NAP firmware is out of date, the device should still respond minimally.  I think you should still be able to get the settings on the device and perform a reset via SBP regardless of the nap firmware. At the very least we should continually print out the error message.

denniszollo, 2015-07-06:
@cbeighley. @fnoble.   I'd like to have this addressed during this sprint (either by me or someone else). 

 I think this needs to be corrected before 1.0.   I think it is really bad from a user's point of view if their device appears bricked. Since we are planning to allow updates to the NAP and STM individually in the GUI, it will become more likely that a user would load on a mismatched set of versions down the road.

cbeighley, 2015-07-07:
Closed by #482

closed 2015-07-07 by cbeighley