swift-nav/piksi_firmware #444: External antenna setting / Flash corruption causes no satellites to be tracked

cbeighley, 2015-06-19:
Customer reported a few issues that seem related and may be caused by the flash corruption issue that was fixed in #433.

* On v0.14, sometimes when Piksi starts up it continuously prints "No Channels Found :(" and doesn't track any satellites. Erasing flash seems to fix the problem. Customer thinks this may have to do with the external antenna.
* Had 5 devices (that were previously working? not super clear), which have suddenly stopped working exciting 1) above.
* Sometimes on v0.17, won't track any satellites. Under Console/Settings, switching Antenna Mode to Patch and saving, and then switching Antenna Mode to External and saving fixes the problem.

cbeighley, 2015-06-22:
Have two Piksies with v0.14 from customer that demonstrate this issue, will investigate using these.

denniszollo, 2015-06-22:
I feel like this might have something to do with renaming / adding some settings