swift-nav/piksi_firmware #475: Handle time-of-week rollovers correctly

henryhallam, 2015-07-05:
There are a few places where we assume that the ephemeris and channel TOW are referred to the same GPS week.  We should go over all the code and ensure correct handling of all conditions, e.g.

 - TOE is week 222 + 6.9 days, tracking TOW is week 223 + 0.1 days
 - TOE is week 223 + 0.1 days, tracking TOW is week 222 + 6.9 days
 - TOE is week 222 + 0.1 days, tracking TOW is week 222 + 6.9 days or week 223 + 0.2 days (e.g. if receiver running for a week and didn't see one sat for most of that time - perhaps we should explicitly invalidate stale ephems)

This is distinct from the issue of making sure we have a consistent and sensible policy for dealing with 1024-week rollovers.

ljbade, 2016-01-07:
@henryhallam I have been working on improving and fixing bugs with time handling in libswiftnav, hopefully I can fix this issue too

ljbade, 2016-01-07:
@henryhallam What is relation between this and https://github.com/swift-nav/libswiftnav/pull/200?

closed 2016-10-13 by ljbade