swift-nav/piksi_firmware #477: Overhaul nav bit processing

henryhallam, 2015-07-06:
Rationales:
 - What kind of crazy asshole has two thumbs and decided it would be a good idea to encode the nav bit polarity as negative vs positive subframe_start_index? ;)
 - Every word (0.6 seconds) is an opportunity to detect (via parity) a half-cycle slip and correct or at least report it.  Presently, we only detect those slips in nav processing via the TLM/HOW once every 6 seconds, and an even number of slips would cancel and be undetected.  It would be better to check the parity of each word as it comes in, so we can detect a slip this way as fast as possible.
 - We aren't checking parity on the TLM/HOW; this may be responsible for some of the rare wildly-incorrect channel TOWs.
 - Having a separate thread for decoding the nav msg is unnecessary (the field extraction is fast enough to be done in the tracking loop update context) and adds to complexity, leading to concurrency crap like the nav bit overruns.

ljbade, 2016-01-07:
@jacobmcnamee Does any of your improvements in libswiftnav bit decoding help here?

ljbade, 2016-01-07:
Also it is amusing that some of the things @henryhallam noted here I recall seeing way back in 2014...

jacobmcnamee, 2016-01-07:
We will need a separate decoding thread when other constellations with more complicated (read: Viterbi) decoding requirements are implemented. I'm currently working on separating tracking and decoding completely but am making minimal changes to the actual decoding algorithms.

ljbade, 2016-01-07:
@jacobmcnamee, ah cool, i might have a look at implementing some of the suggestions around parity

jacobmcnamee, 2016-01-07:
Hah also get rid of that negative index nonsense.

ljbade, 2016-01-08:
Another relevant issue https://github.com/swift-nav/piksi_firmware/issues/357