swift-nav/piksi_firmware #502: MsgPrint INFO/WARNING newline termination issues

mookerji, 2015-07-16:
On testing for v0.19-rc0, seeing occurrences of multiple message print contents in the same payload field. See: 

```
'INFO: IAR: 4WARNING: dgnss_baseline returned error: -1\n',
       'INFO: IAR: 4WARNING: dgnss_bae',
       'INFO: IAR: 2WARNING: dgnss_baseline returned error: -1\n'
```

but in libswiftnav, 
```
src/ambiguity_test.c:764:  log_info("IAR: %"PRIu32" hypotheses before projection\n", memory_pool_n_allocated(amb_test->pool));
src/ambiguity_test.c:769:  log_info("IAR: updates to %"PRIu32"\n", memory_pool_n_allocated(amb_test->pool));
src/ambiguity_test.c:1004:  log_info("IAR: updates to %"PRIu32"\n", num_hyps);
src/ambiguity_test.c:1711:  log_info("IAR: %"PRIu32" hypotheses before inclusion\n", memory_pool_n_allocated(amb_test->pool));
src/ambiguity_test.c:1719:  log_info("IAR: updates to %"PRIu32"\n", memory_pool_n_allocated(amb_test->pool));
```

@henryhallam @cbeighley @fnoble @gsmcmullin @ungureanuvladvictor

henryhallam, 2015-07-16:
ok so probably not newline-related.  In which thread is each message printed?

mookerji, 2015-07-16:
Yeah, so `output_baseline` in src/solution.c is called by both the solution thread and the time matched obs thread, and I think the IAR message occurs in one of those as well.

mookerji, 2015-07-16:
btw, this is on the field testing log from 07-13-tests/Test_Mon_07_13_2015_151018_70.log.json