swift-nav/piksi_firmware #593: Excessive unnecessary calls to printf via sid_to_string

jacobmcnamee, 2016-01-07:
Preemptively printing out the sid string at the top of a function is wasteful:

```
char buf[SID_STR_LEN_MAX];
sid_to_string(buf, sizeof(buf), ch->sid);

if (<unlikely condition>)
  log("error: %s", buf);
```

ljbade, 2016-01-07:
@jacobmcnamee I did this to avoid copy/pasting code and to use principles of reuse.

If you think the performance overhead is significant then we can go back to copying the code in several places.

jacobmcnamee, 2016-01-08:
Yeah, I understand the reasoning. But that's a lot of printfs that we rarely need. Especially in places like the tracking thread.

ljbade, 2016-01-08:
@jacobmcnamee Yeah OK move the extra printfs out of the high perf threads like tracking. For other low priority threads I think they can be left.

jacobmcnamee, 2016-01-08:
How about
```
#define _LOG_SID(func, sid, format, ...) { \
  char sid_str[SID_STR_LEN_MAX]; \
  sid_to_string(sid_str, sizeof(sid_str), sid); \
  log_error("%s" format, sid_str, ##__VA_ARGS__); \
}

#define log_warn_sid(sid, format, ...) \
  _LOG_SID(log_warn, sid, format, ##__VA_ARGS__)
```

Then you can do
```
log_warn_sid(sid, "something bad happened");
```
or
```
log_warn_sid(sid, "return code: %d", code);
```

ljbade, 2016-01-09:
Hmm yeah that could work. Would prefer to not use a macro tho as they can be misleading as it looks like a real function.

However perhaps http://stackoverflow.com/questions/24814570/can-i-add-new-format-specifier-type-to-c-printf-in-linux-platform might work...

jacobmcnamee, 2016-04-04:
swift-nav/libswiftnav#319
#632

closed 2016-04-04 by jacobmcnamee